stevieb9/rpi-wiringpi #19: RPi::WiringPi::LCD has to call WiringPi::API::setup()

gflohr, 2017-03-01:
In order to make RPi::WiringPi::LCD work as expected in a standalone script, I had to call WiringPi::API::setup() explicitely, see below:

#! /usr/bin/env perl

use strict;

use RPi::WiringPi::LCD;

my $lcd = RPi::WiringPi::LCD->new;

my %lcd_args = (
    rows => 2, cols => 16, bits => 4, rs  => 2, strb => 3,
    d0   => 6, d1   => 5, d2   => 4, d3   => 1,
    d4   => 0, d5   => 0, d6   => 0, d7   => 0,
);

WiringPi::API::setup();  # FIXME!

$lcd->init(%lcd_args);
$lcd->clear;
$lcd->position(0, 1,);
$lcd->print("It works!");

gflohr, 2017-03-01:
I tried to instantiate the lcd object as recommended with the factory method instead:

    my $lcd = RPi::WiringPi->new->lcd;

Now the script does not work at all, with or without the explicit call to the setup() method.

Am I missing something?

gflohr, 2017-03-01:
Okay, got it now. ;) Depending on how you instantiate the object, you have to change the pin numbers.  This version now works with the factory method and the same wiring:

    my %lcd_args = (
        rows => 2, cols => 16, bits => 4, rs => 27, strb => 22,
        d0 => 25, d1 => 24, d2 => 23, d3 => 18,
        d4 => 0, d5 => 0, d6 => 0, d7 => 0,
    );

This is then rather a documentation issue than a bug.

stevieb9, 2017-03-01:
Thanks. I'll consider what to do here. I remember that there was a design
reason for this at one time, but I believe the issues that caused me to go
that route are now fixed. I must have missed making notes in the docs.

stevieb9, 2017-03-02:
Reopening, as though I haven't had time to thoroughly go through the exact issues, differences between instantiations do disturb me. Often, I read my own documentation so I can remember how to use my own software, so this discrepancy needs to be addressed.

stevieb9, 2017-03-03:
Self note:

The fix is going to have to be applied to all internal sub modules. This goes along with (closed) #16 (remove all setup routines less gpio).

In that ticket, we kind of hid the fact we still allow all setup routines (but I did notice in FAQ they are all still mentioned).

What does work is `$lcd->setup_gpio;`. 

The fix may be just a documentation update. I'm hesitant to just call `setup()` blindly, as with recent changes in wiringPi, this may lead to a file handle exhaustion issue.

Consider updating the check in `::Util`, and call `setup()` if it has not been called yet (along with doc updates).

In testing.

stevieb9, 2017-04-28:
The LCD module was exported out of `RPi::WiringPi` as a standalone distribution `RPi::LCD`. We just import it into here now.

The setup routine was added into the separate dist, with a check to only run if it hasn't been already called.

closed 2017-04-28 by stevieb9