stevieb9/rpi-wiringpi #157: setting mode PWM_OUT frezze my Rapspberry Pi 3

guillepo, 2019-09-25:
Dear Stevie, 
thankyou very much for this package.
I am traying to use RPi::WiringPi and it did not work
for setting a pin to pwm case. I think that the problem is
at this line: $pin->mode(PWM_OUT);    
in a simple code like:
#!/usr/bin/env perl                                                       
use RPi::WiringPi;
use RPi::Const qw(:all);
use feature qw(say);
                                            
my $pi = RPi::WiringPi->new;

# pin                                                                     
my $pin = $pi->pin(18);
        
say PWM_OUT;
$pin->mode(PWM_OUT);                                                     
say "pin mode:\t",$pin->mode;
                                                      
$pi->cleanup;

stevieb9, 2019-09-25:
Thank you for reporting. Can you please let me know what the problem is and
any errors you're receiving? The only thing I can think of is that you're
not running under 'sudo', but that should produce a warning. This works for
me:

```
use RPi::WiringPi;
use RPi::Const qw(:all);
use feature qw(say);

my $pi = RPi::WiringPi->new;

my $pin = $pi->pin(18);

say "PWM_OUT: " . PWM_OUT;

$pin->mode(PWM_OUT);
say "pin mode:\t",$pin->mode;

$pin->mode(OUTPUT);
say "pin mode:\t", $pin->mode;

$pin->mode(INPUT);
say "pin mode:\t", $pin->mode;

$pi->cleanup;
```

Output:

```
PWM_OUT: 2
pin mode: 2
pin mode: 1
pin mode: 0
```

stevieb9, 2019-09-25:
Also, please let me know the following output:

`perl -MRPi::WiringPi -E 'say $RPi::WiringPi::VERSION'`

...and

`gpio -v`

guillepo, 2019-09-26:
Thank you Steve,
After you first reply I realize that pwm requires some internal
sudo privileges as it is mentioned in FAQ web page.
Now, after appropriate adjustment os sudoers path seems that
it is working.

Anyway,  the version that I am using of WiringPi: 2.3633
And, it is my RPi version:
Copyright (c) 2012-2017 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Embest
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.
=================

Regards

stevieb9, 2019-09-26:
I'm very glad you got it working. Let me know if you have any further problems. I'll update the documentation to be a bit more clear.

closed 2019-09-26 by stevieb9